rlogiacco/BatterySense #2: Unsteady reading result

nandakrpp, 2016-04-06:
Hello,

I'm using the library with Arduino mega to measure 2 of 6AA battery packs wired in parallel. Measured with a digital voltmeter I got 8910 millivolts which is different from the result read by the library. Here is the circuit diagram of my setup: [voltmeter_batterySenseLib_0_2_schem](https://goo.gl/photos/iceLspoox5rsgVTB7) and below is the result captured from Arduino serial monitor:

```
Battery voltage is 4960 (100)
Battery voltage is 60496 (0)
Battery voltage is 60496 (100)
Battery voltage is 60496 (100)
Battery voltage is 4960 (0)
Battery voltage is 14960 (100)
Battery voltage is 50496 (100)
Battery voltage is 60496 (100)
Battery voltage is 4960 (0)
Battery voltage is 60496 (100)
Battery voltage is 14960 (0)
Battery voltage is 57104 (100)
Battery voltage is 4960 (100)
Battery voltage is 60496 (100)
Battery voltage is 4960 (100)
Battery voltage is 4960 (100)
Battery voltage is 4960 (100)
Battery voltage is 4960 (100)
Battery voltage is 14960 (100)
Battery voltage is 40496 (100)
Battery voltage is 14960 (0)
Battery voltage is 14960 (100)
Battery voltage is 4960 (100)
Battery voltage is 4960 (0)
Battery voltage is 4960 (100)
Battery voltage is 4960 (0)
Battery voltage is 60496 (100)
Battery voltage is 14960 (100)
Battery voltage is 4960 (0)
Battery voltage is 60496 (0)
Battery voltage is 14960 (100)
Battery voltage is 50496 (0)
Battery voltage is 14960 (0)
Battery voltage is 4960 (0)
Battery voltage is 14960 (100)
Battery voltage is 4960 (100)
Battery voltage is 60496 (0)
Battery voltage is 4960 (100)
Battery voltage is 14960 (100)
Battery voltage is 4960 (100)
Battery voltage is 14960 (0)
Battery voltage is 60496 (100)
Battery voltage is 14960 (0)
Battery voltage is 4960 (100)
```
I also did a calibration using EEPROMCalibration code but having no improvement in the reading result.
Any pointers are appreciated.

Thanks.

rlogiacco, 2016-04-08:
Are you using Q1 to enable the sensing circuit only?
Can you please post your code?
I've already received some indication my calculations could have been wrong so your feedback can help me solve this issue.

rlogiacco, 2016-04-09:
This should now be fixed within the latest version, thankfully to @nunonunes : I would appreciate if you can please confirm it's now working as expected considering all the values other than battery level should still be expressed in mV.

Thanks

closed 2016-04-22 by rlogiacco