rlogiacco/BatterySense #3: higher-on-demand circuit

mesmerise, 2016-06-16:
Hi there,

big up & thx for the library, it works very well!

There seems to be a minor issue with the suggested circuit for the higher-on-demand scenario. Correct me if I am wrong, but I believe the n-channel MOSFET in circuit diagram is connected wrongly: Source should be connected to GND and drain should be connected to R2 and not vice versa as shown in higher-on-demand.png. 

Best,
Andreas

rlogiacco, 2016-06-16:
Hi Andreas, good catch, you are definitely right!
Would you like to submit a pull request with an updated circuit diagram for that?

mesmerise, 2016-06-16:
Thanks mate, I’m fairly new to Github - I tried to create a pull request with an updated diagram - hope it worked..

rlogiacco, 2016-06-16:
It did work indeed... Can you remove the additional _V3_ part at the end of the file name?

mesmerise, 2016-06-16:
I couldn’t change the filename directly, but I uploaded a renamed version just a few minutes ago.

rlogiacco, 2016-06-16:
pull request accepted: thank you!

closed 2016-06-16 by rlogiacco