rlogiacco/BatterySense #7: What to do if voltage supplied is a little higher than the requred voltage.?

blaine12100, 2017-09-22:
Hi

I will be using the library to measure the voltage of a battery rated at 3.7v and my circuit would be supplied with 3.3v.For this scenario,i don't need a voltage divider circuit.So would it be ok if i made the default voltage divider ratio as 1(No effect on output,i think) and the default reference voltage as 3330 Milli volts in Battery.h file?

Do let me know if i am correct.

rlogiacco, 2017-09-22:
This is not an issue, but a question.
Nonetheless, you want to measure your battery, not the voltage output of your MCU, right?
If, in your case, it is OK to report a 100% full battery between 3.7V and 3.3V, and then start reporting 90% when battery level reaches 3.25V, 80% at 3.2V, etc... Than it's fine with 3300 as setup value.
To me, that would be wrong: my circuit might operate at 3.3V, but my battery starts at 3.7V and when it reaches 3.6V I want to know I'm left with 95% battery... 
Anyway, what type of battery do you use? Because I don't know of any battery operating at 3.7V max... If it's a Li-Ion you are getting this pretty wrong...

blaine12100, 2017-09-23:
I am actually waiting for the battery to arrive.And yes i would like to just report the battery level not the output from the micro controller.So what changes would i need to make in order to correct the wrong part as mentioned by you.Do you mean to say that i need to keep the reference voltage also as 3.7v and the dividing ratio as 1 is ok?

rlogiacco, 2017-09-23:
No, I'm saying you should use a voltage divider and, supposing yours is a Li-Ion battery (4.2V - 3.7V) you'll need to use a voltage divider with a ratio between `1.3` and `2`. 
For example, you can use a 4.7kΩ (R1) and a 10kΩ (R2) and set a ratio of `1.47` allowing you to measure batteries with a maximum voltage of 4.85V, well within the swing of a Li-Ion.

I've added a few usage examples to the README: please refer to that.

closed 2017-09-23 by rlogiacco