rlogiacco/BatterySense #8: [Question] Sense pin configuration with on-demand scenario

papylhomme, 2017-11-26:
Hello,

I'm wondering what happens regarding the sense pin when Q1 is switched off ? According to calculation the voltage supplied to the pin is much higher than when Q1 is on, and it will consume current through R1.

Do I need to switch the sense pin to output or something ?

papylhomme, 2017-11-26:
OK for the voltage divider and its output limited to under 3.3V, and regarding my concern for the sense pin with Q1 turned off, everything is kept fine due 100MΩ to the internal resistance.

So if I understood everything correctly, the Arduino won't see the 4.5v, EXCEPT if I try to read the value WITHOUT switching Q1 on, in which case I'll burn the pin and maybe the arduino ? I saw the library takes care of this in the `voltage` method, just checking my understanding of the related electronics.

rlogiacco, 2017-11-26:
Yes, you risk your sense PIN functionality (Arduino should be safe, due to internal diode protection) only if you read without enabling the mosfet. With the current limiting protection of the 100MΩ resistor also the pin might remain intact even in that situation: the current allowed to pass through should not harm the pin anyway, but better not to risk.

If you are concerned about all of the above, you can switch to a p-channel mosfet configuration, but you'll have to also add a NPN transistor to drive the mosfet from the higher voltage.

papylhomme, 2017-11-26:
Honestly I'm more concerned about burning everything by misunderstanding what I'm doing ! I used to think about transistors as simple switches and now I'm in over my head with NPN/PNP and n/p channel MOSFETs. 

Anyway, thanks for the explanation and for the library, I'll try this using the higher on-demand configuration (almost) confidently.

closed 2017-11-26 by papylhomme