rlogiacco/BatterySense #17: MOSFET and activate measurment with digital PIN

rlogiacco, 2018-11-16:
I'm interested in the chapter "Higher than 5V, activated on demand".

Could I increase the R1 and R2 resistances in order to decrease the current, i.e. `10k` ohm?

Reading the values of the scope, I saw the MOSFET is running in linear mode. Isn't it better if the MOSFET works in saturation mode? Because the inner resistance Rds is a tiny value and, opposite to that, in linear mode, the inner resistance must be greater than in saturation mode.

I've changed your circuit to change pMOSFET for nMOSFET. This worked fine too, but not exactly the same values. Which of them is the better choice?

In addition, I'm very confused with the Threshold voltage concept. I know this is the minimum voltage to close the MOSFET. I've read the voltage to saturate a usual MOSFET must be `10V`. I suppose than Threshold voltage is the minum voltage to enter in a linear mode, but it's very interesting to arrive the voltage to pass a saturation mode. Is it correct? I've also read we can use a differrent MOSFET, the logic level MOSFET that can operate at 5v. Is it correct too?

rlogiacco, 2018-11-16:
Given the mosfet should be run in saturation mode to reduce the RDS and also the heat it generates, the resistor values can be changed at your convenience, according to the voltage divider ratio and the current consumption.

With regards to the p-channel vs n-channel, what matters is the voltage difference, not the absolute value. Try to turn on a n-channel mosfet for a 25V line using 5V: that's never going to work... It really depends on the circuit and the voltages involved, but I'm no expert on the subject.

If you are encountering those type of problems you probably want to switch to a bipolar transistor: being current operated the voltages have a lower impact, but once again, it really depends on the actual circuit.

Please also note there is a problem when using n-channel mosfets (which is the reason I suggest a p-channel mosfet) and it was an issue (#3) raised by a user of the library.

closed 2018-11-16 by rlogiacco

mrosh, 2019-01-15:
Hello, can you please write me what kind of mosfet and transistor I should buy.
Thank you

rlogiacco, 2019-01-15:
There's no one-off solution, it depends on the voltages you are dealing with (battery vs controller).

As a general rule of thumb, assuming you are dealing with a `9-12V` battery and your voltage divider has a total reistance (R1 + R2) not less than `2kΩ` any _logic level P-MOSFET_ will do (eq. `IRL5602`, `LP0701N3` and `5LP01SP` are good candidates).

The definition of _logic level MOSFET_ can be roughly translated to "having an activation threshold voltage under 5V": please refer to the many available sources out there,  like https://forum.arduino.cc/index.php?topic=456591.0

If you don't have a logic level P-MOSFET at hand, than you can get away using any P-MOSFET (non _logic level_) and a BJT transistor, like the widely adopted `2N2222` NPN transistor: just don't forget to add a series resistor to the transistor's base to limit the current.