rlogiacco/BatterySense #19: Does it work with 12V sealed lead acid (SLA) batteries?

jdwolk, 2019-01-31:
Hi!

Forgive me if this is a stupid question; I'm very green when it comes to EE / microcontrollers and I'm just wrapping my head around the basics.

Will this project be able to monitor 12V sealed lead acid (SLA) batteries?

Thanks!

rlogiacco, 2019-01-31:
The library, per se, has no limitations in those terms, you must be careful though on the electronics side in terms of voltages: from a quick search it appears those batteries reach a maximum of `13V` (fully charged) and a minimum of `10.5V` (complete depleted), but you must confirm those values.

Also, looking at the typical discharge curve, those batteries seem to have a very different character, so you either go for the `linear` or define your own approximation function.

Other than that I don't see a reason why the library shouldn't work, even if I've never implemented anything for those batteries myself: drop us a line and share your results.

jdwolk, 2019-02-01:
Thanks for a quick response! Closing now + will report back if I'm able to get things to work

closed 2019-02-01 by jdwolk