rlogiacco/BatterySense #21: measuring up to 48v baterry (10 AH)

ibnukhak, 2019-05-28:
can this be used to sensor the battery above 48v with 10 amphere hour? if which can be changed?

rlogiacco, 2019-05-28:
I don't see a reason why it shouldn't work, provided you configure it properly and you use a control circuit fit for your battery.

closed 2019-05-28 by rlogiacco

ibnukhak, 2019-05-28:
do I need to replace the voltage divider resistor ?, so that I get 0 - 5 volts at the point connected with Arduino, my battery is lithium polymer, sometimes the voltage is above 48 V and has 1O Ampherehour, thks before

rlogiacco, 2019-05-28:
I have spent hours writing the project README: please read it extensively and accurately, then, if you still have doubts, I will be happy to help answering to questions that are not in the form _“give me the parameters for my battery”_