rlogiacco/BatterySense #22: Analog pin is behind multiplexer.

bhattvishal, 2019-06-14:
I have just a single analog pin, and I have connected to a multiuplexer to get more analog pins. Is there any way to just pass the value instead of sense pin?

Mingyu-Kim, 2019-06-28:
```
Battery::Battery(uint16_t minVoltage, uint16_t maxVoltage) {
	this->minVoltage = minVoltage;
	this->maxVoltage = maxVoltage;
}

void Battery::begin(mapFn_t mapFunction) {
	this->mapFunction = mapFunction ? mapFunction : &linear;
}
```

I have added following functions in the cpp/h to create battery class without sense pin
and used uint8_t Battery::level(uint16_t voltage) in mv to get level.

rlogiacco, 2019-07-02:
If all you want is to map a sensed voltage to a battery percentage then you donàt need to instantiate the `Battery` class: just use any of the static functions provided: 

- sigmoidal(uint16_t voltage, uint16_t minVoltage, uint16_t maxVoltage) 
- asigmoidal(uint16_t voltage, uint16_t minVoltage, uint16_t maxVoltage)
- linear(uint16_t voltage, uint16_t minVoltage, uint16_t maxVoltage)

You only have to convert the value you sensed to a voltage. What suggested by @Mingyu-Kim only wastes additional memory without any benefit whatsoever.

closed 2019-07-02 by rlogiacco