rlogiacco/BatterySense #23: Require information on how to monitor individual cells for Voltage in a 4 Pack (3.7V each) Li-Ion Battery

SowmyaGN, 2019-07-11:
I am working on creating  miniature Battery Management System (4 Cells : Li_ion : Each cell 3.7V 2600 mAH ) for my student project. This project is very much close to what I am trying to accomplish. 

When I looked into the project, I observed that there are libraries to read the battery voltage as a whole. Is there any way to read the individual cell voltage. 
Your help is greatly appreciated in this regard.

rlogiacco, 2019-07-11:
There is no software limitation preventing you to have one instance of the `Battery` class per each cell plus one for your battery pack, but I wonder how you would tap into each of the cells and measure their voltage: you require a circuit capable of returning a voltage referred to GND starting from a voltage referred to above-GND...

SowmyaGN, 2019-07-12:
Thanks, . I was thinking i will have individual connections to the battery and common ground.  - Will that not work

rlogiacco, 2019-07-14:
If the cells are all in parallel yes, it will work, but if the cells are in series, I can't see a simple way to have a common ground: you are either connected to the pack ground or tap into any virtual ground of the battery pack... You can still create separate circuits, but to my humble opinion, this is usually overkill and much more expensive than generally thought.
Can you share a circuit? I can suggest using https://www.falstad.com/circuit/circuitjs.html and share here a link created via _File > Export as link... > Create Short URL_

rlogiacco, 2020-02-21:
closing due to lack of activity

closed 2020-02-21 by rlogiacco