rlogiacco/BatterySense #24: Using library with 18650 and Attiny84

diogoribeiro09, 2020-03-26:
Hey,
I'm trying to use the library with an Attiny84 powered by a single 18650. The attiny works from 1.8-5.5v although I want it to go no lower than 3v (I'm also using an nrf24). The 18650 outputs 4.2 max charge and under 3v is a no go. I am using a voltage divider with R1 = 4.7 and R2 = 10k.  I am getting a zero output if I call the battery.level() function. 
Thoughts?

Thanks for work.

rlogiacco, 2020-03-26:
Please post your code, in particular what you have used to initialize the `battery` object, along with the wiring info (which pin is your sense pin and which is the enabling pin?) and the usage model you wish to use in your case (always active?)

In general, if the `level` function is returning ’0’ you have probably initialized the object wrongly because the measured voltage is constantly below the minimum allowed voltage.

diogoribeiro09, 2020-03-26:
I want it to be always active although I may change it in the future. My sense pin is the A0 of the Attiny84. I have no enabling pin.

The object is initialized as follows:
`//Battery battery(3000, 4300, A0);`

On my setup function:
`// battery.begin(3300, 1.47, &sigmoidal);`

And that's all I'm using apart from the _battery.level()_ function which results are then sent over the NRF (communication is working well).
I'm simulating the circuit by connecting it to a Digital Lab, monitoring the VIN and trying to make sense of the result.

rlogiacco, 2020-03-26:
I suggest you first start testing the `voltage` function returns the correct input voltage in millivolts: that is the foundation for the `level` function computation.
If you set a `3300` reference voltage your A0 pin analog value will swing between `0` and `1023` with the highest value representing `3.3 V`: this, along with the ~1.5 value for the voltage multiplier means a 1023 analog value on the A0 pin equals roughly to `4.85 V` on the battery.
Now, the `voltage` function should return a value of 4200 (the voltage measured at the sense pin after applying the required mapping to volts and multiplication due to voltage divider) if you have 4.2 V at the high side of the voltage divider.

Is it possible you are also using A0 or 0 or PCINT0 for some other goal?

Also, and again, please post your schematics and full code: I'm not willing to continue guessing any possible culprit I can imagine.

diogoribeiro09, 2020-03-27:
I figured it out during the evening. I had three wires that were producing an inconsistent signal. It is working right now.  Thanks for the work

closed 2020-03-27 by diogoribeiro09