rlogiacco/BatterySense #28: 2x18650 in series

T3N0R10, 2020-07-17:
I've got 2 Li-Ion cells 18650 in series as supply on an arduino nano v3 and I  would want to know if I can sense one of them directly (the one with the negative connected to gnd and a line to an analog pin) and after sense both of them with a voltage divider( to a different analog pin. Then make the maths to known approximately the votage of both. D'you think it's possible? I want to make the circuit "on demand" that you post (once for each) and sense it separataly, at different times, to avoid shorts or some. If you think that there are a better way, please tell me. Sorry about my english.
Thanks Roberto!!!

rlogiacco, 2020-07-17:
What you are describing is certainly possible, the circuit is probably going to be two copies of the `on demand` version running in parallel and you will need two separate instances of the library. Our recent versions of the library should also help you substantially to determine the _percentage_ because we have externalized the function so you can call it with your _adjusted values_.
But... are you 100% sure you need to measure the two batteries separately? If they are connected in series they will quickly balance...

closed 2020-07-17 by rlogiacco

T3N0R10, 2020-07-17:
You realized me that the batteries will "auto balance" in a few minutes. That simplify me A LOT my circuit. In propouse, wich NPN transistor Q1, an p-channel MOSFET Q2 do you recomend to make the "on demand" circuit for my dual 18650 (they are 3000mah and 20A instant discharge)? It's a smd board and I would like to test it every 5 or 10 seconds. Me and basic electronic don't know to much yet.
Thanks again Roberto!!!
BTW, your library Is amazing!

rlogiacco, 2020-07-18:
Nice to have helped. 
Mosfet selection depends on too many factors for me to pick up one for you: sorry.