rlogiacco/BatterySense #29: Arduino MKR WiFi 1010 Battery Level

philipdbrown, 2020-07-23:
Hey,
Thanks for this library.  However, I'm having issues obtaining the battery level.  It always reads 100.  I have a Li-Po 3.7v battery connected to the Arduino through the on-board JST connector.  Any ideas on what issues I may be having?  Thanks in advance.

```
Battery battery = Battery(3000, 4200, ADC_BATTERY);
...
battery.begin(3300, 1.47);
...
battery.level()
```

rlogiacco, 2020-07-23:
Sorry mate, that's a question for the Arduino forum, and I advise to:

* provide the full source code
* provide a circuit diagram

None of us has a crystal ball :wink:

If you get a solution and you wish to share it here I'll update the issue label from `invalid` to `question`.

closed 2020-07-23 by rlogiacco

tangram67, 2021-01-17:
Although this thread is closed I will comment the request. Maybe the answer is helpful:
The level is read to 100% because the MKR analog uses an internal reference voltage at 2.23 Volt. The voltage of the battery is much higher, so you should use the 3.3 Volt on the AREF pin. Also set the reference voltage to external source:
  // Set ADC parameters
  analogReference(AR_INTERNAL);
  analogReadResolution(10);
Have not tested this, but should work.

rlogiacco, 2021-01-19:
Tanks a million mate for your answer: definitely I didn't know about the different reference.
I would like to add another option for those who might incur into the same issue: use the internal reference but use a voltage divider to get into the correct range.

tangram67, 2021-01-19:
Great idea to use this as an option! Here is an example for a Arduino MKR NB 1500: The on-board the divider is 1.275 (the resistors are 1.2M and 0.33M). So the analog input range is 3.3 * 1.275 = 4.2 Volt and that fits to the max. loading voltage of a Lithium battery ;-)