rlogiacco/BatterySense #30: Battery level with additional load

Rimbaldo, 2020-08-18:
My circuit is powered by a 7,4V battery. The battery is connected to my Leonardo VIN. When I turn on a 3W led, with it's driver circuit  powered directly by the battery, and enabled/disabled and controlled it's intensity by the Arduino, the voltage level in the battery drops. 

Do you have any functions that allow to declare something like a "load voltage drop" in the battery initialization options? Something like if a load is turned on (detected by an input pin,, or pwm different than 0), it then corrects the voltage readings? 
 
Regards, Rodrigo

rlogiacco, 2020-08-19:
Hi @Rimbaldo ,
No, the library doesn’t provide anything like that, but you can change the library to compensate and normalize the readings on your needs...

rlogiacco, 2020-08-19:
BTW, you might want to submit a PR with your changes and I might incorporate that feature in the library, if like to contribute it

Rimbaldo, 2020-08-20:
> incorporate that feature in the library, if you like to contribute it

Ok, I'll try later then... Thanks!!

closed 2020-12-02 by rlogiacco