rlogiacco/BatterySense #43: Higher voltage than expected

Dans-Droids, 2021-12-08:
### Discussed in https://github.com/rlogiacco/BatterySense/discussions/42

<div type='discussions-op-text'>

<sup>Originally posted by **Dans-Droids** December  6, 2021</sup>
Hi, thank you for making this brilliant library!

I am finding that on my STM32F103C8, powered by a single Li-Po through a 5v3a regulator with the sense pin on PA1, the returned voltage is higher than it should be.

I have found that with the value of 5000 in the following, the recorded voltage is around 7680:

battery.begin(5000, 1.47, &sigmoidal);

- if i change the 5000 to 3300, this causes the recorded voltage to be around 4850 .

Do i need to reference the 1.1v Analog Voltage of the pins?
I am planning to use the ADCTouchSensor library on pin PA6 - this requires the Analog Reference to be floating... Would this interfere with this BatterySense library??

Thank you for your help!</div>

Dans-Droids, 2021-12-10:
Thank you for the response. I forgot to mention that the SensePin is directly connected to the Li-Po batterie's +ve...
Am I correct in expecting the SensePin to be the same voltage as the Li-Po?

If this is the case, can you help me understand why the value would be close to 5v - when it should be 3.8-4.2V?
Also I noticed that the value does not change as the battery drains - even though the code is in the loop in an otherwise empty sketch?

rlogiacco, 2021-12-10:
Please attach a schematics of your circuit.

Dans-Droids, 2021-12-10:
it is exactly as pictured in the "lesser than 5v with voltage booster"
[https://github.com/rlogiacco/BatterySense#lesser-than-5v-with-voltage-booster](https://github.com/rlogiacco/BatterySense#lesser-than-5v-with-voltage-booster)

rlogiacco, 2022-01-18:
Is your device a 5V or a 3.3V device? Is the battery a single cell LiPo or a two cells LiPo?

Unless you are on a 5V device and a single cell LiPo I don't see how you can measure the battery voltage because it would be above the sensing limits.

The MCU you are referring to (STM32F103C8) is a 3.3V device, meaning the highest voltage it can sense is 3.3V and, if powered from a 5V source, such 5V is stepped down to 3.3V by an onboard regulator.

The LiPo you are measuring, assuming it is a single cell, has a voltage ranging from 3.7 to 4.2 and to measure it you need a voltage divider.

Dans-Droids, 2022-02-17:
Hi rlogiacco,

Sorry for the delay in responding, I have been working on other projects and see your reply..

Yes you are correct, it is a single cell LiPo giving 3.8V - 4.2V. This goes to a dc-dc converter that gives constant 5V3A output to power the STM32F103C8 and other devices..

I shall connect up a voltage divider and see what happens.

Thank you for your help

closed 2022-07-12 by rlogiacco