rlogiacco/BatterySense #47: Which voltage divider for 3.3v board?

fyrus7, 2024-01-11:
The main readme suggest voltage divider like this for 3.3v board

Battery + ----
                     |
                     |
                     |
                    R1 (4.7kΩ)
                     |
                     |
                     |
                     |--------- A0
                     |
                     |
                     |
                    R2 (10kΩ)
                     |
                     |
                     |
Battery - -----

How to connnect this actually? Is it like this?

Battery + ---------- VCC
                     |
                     |
                     |
                    R1 (4.7kΩ)
                     |
                     |
                     |
                     |--------- A0
                     |
                     |
                     |
                    R2 (10kΩ)
                     |
                     |
                     |
Battery - ---------- GND

rlogiacco, 2024-01-12:
No, the positive side of the battery should go into **Vin** (to be further regulated to 3.3V), the negative into GND.
If your board doesn't have an onboard regulator you should buy or build one: a battery cannot provide a stable input voltage.

fyrus7, 2024-01-19:
> No, the positive side of the battery should go into **Vin** (to be further regulated to 3.3V), the negative into GND. If your board doesn't have an onboard regulator you should buy or build one: a battery cannot provide a stable input voltage.

So the divider is from VCC and GND > junction to A0?