rlogiacco/BatterySense #51: Battery voltage reporting too high

entozoon, 2024-07-08:
Hi, I'm surely doing something wrong, but here is my situation in designing a voltage tester for 3S li-ion batteries.

```cpp
#include <Arduino.h>
#include <Battery.h>
// *    +--------+
//  *   |        |
//  *   |       +-+
//  *   |       | | 40k  R1
//  *   |       | |
//  * -----     +-+
//  *  ---       |---------+ A0
//  *   |       +-+
//  *   |       | |
//  *   |       | | 10k  R2
//  *   |       +-+
//  *   |        |
//  *   +--------+---------+ GND
//  **/
const float R1 = 40000.0; // ohms
const float R2 = 10000.0;
const uint16_t battMin = 10500; // millivolts
const uint16_t battMax = 14400;
Battery battery(battMin, battMax, A0);
void setup()
{
  Serial.begin(76800);
  while (!Serial)
    ;
  float ratio = (R1 + R2) / R2;  // 5
  battery.begin(5000, ratio);
}
void loop()
{
  int analogValue = analogRead(A0);
  float batteryVoltage = battery.voltage();
  int batteryLevel = battery.level();
  Serial.print("Analog value: ");
  Serial.print(analogValue);
  Serial.print(", Battery voltage: ");
  Serial.print(batteryVoltage);
  Serial.print(" mV, Battery level: ");
  Serial.print(batteryLevel);
  Serial.println("%");
  delay(200); 
}
```

At 10.5 volts (the minimum, which should report as 0%), the output is similar to:

```Analog value: 675, Battery voltage: 16479.00 mV, Battery level: 100%```

Perhaps I'm misunderstanding how the ohms law ratio works here, but the analog value seems to be safely within 1024 as expected but yeah, the battery voltage stated in mV is way too high

entozoon, 2024-07-08:
Hmm, if I set my `refVoltage` to 3306, which is what I measure across the 3V3 pin and GND (despite powering from 5V USB) then the figures are much closer. Still half a volt out, but I guess it's better than nothing and I can fudge the numbers ..

rlogiacco, 2024-07-11:
Which MCU is your code being run on?
The 5000 value passed as argument at the begin() method is the
microcontroller’s beef value of the internal ADC…

entozoon, 2024-07-12:
Hi @rlogiacco, thanks. I have a [Wemos S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) which is an ESP32-S2FN4R2.
I do not understand how to discovery the correct ref voltage.. :(

rlogiacco, 2024-07-12:
So, your board is a 3.3V board and also the ADC seems to be recommending an external 100nF capacitor: 
[image]

So, you need to change the `begin ` method parameter to `3300` and add the capacitor. I also suggest measuring the real values of the two resistors used for the voltage divider to increase accuracy.

To be clear, the ESP32 (the MCU at the core of your board) has a `1.1V` vRef (specs from the [manufacturer](https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32/api-reference/peripherals/adc.html) state this is just a median value though), this is then converted to a `3.3V` reference by the board (by using a voltage divider, which might be introducing inaccuracy), on top of that you are using an additional voltage divider... Sadly the board you are using is not really the best choice for your goal if accuracy is key.

BTW, I do not own the board you are using, thus I’m only reading specifications found on the internet.

closed 2026-05-23 by rlogiacco